DFKI-NI/seerep #128: Update seerep-msgs README

mhoellmann, 2022-07-26:
The Readme of seerep-msgs is outdated and needs an update.

- create a documentation that fits flatbuffer and protobuf
- also document core-msgs. As Doxygen code docu because it is c-code?

https://github.com/agri-gaia/seerep/blob/f4696ae1a3f1e21b788e1d1f62fdae9844099958/seerep-msgs/README.md?plain=1#L1-L410

@jarkenau: any suggestions how to integrate this in MkDocs or Doxygen?

jarkenau, 2022-07-28:
Since Doxygen has Markdown support, we can use the config as we did with the main page. Basically it just has to be included here: 
https://github.com/agri-gaia/seerep/blob/c0214ed5e015eb49b620bc0e2ce4f5049e6138ad/Doxyfile#L832
or if we want to include all `READMEs` from the `seerep/...` packages we could use a file pattern 
https://github.com/agri-gaia/seerep/blob/c0214ed5e015eb49b620bc0e2ce4f5049e6138ad/Doxyfile#L859-L862 

I tested it briefly and noticed that the links are not working in Doxygen, so we probably need some further examination there.

jarkenau, 2022-07-28:
For MkDocs things could get more complicated, as it expects the documentation files to be in the `docs_dir`. A symlink or copying it in there during the GitHub workflow would of course work, but those are more or less dirty solutions. 

But as you already pointed out it's probably best to include the messages documentation in Doxygen, because mostly devs would care about that.

jarkenau, 2022-07-28:
Searched a bit, if we want to include it in MkDocs we could use [this](https://github.com/athackst/mkdocs-simple-plugin) plugin.

closed 2023-11-22 by Mark-Niemeyer