DFKI-NI/seerep #151: Create debian packages

mhoellmann, 2022-08-11:
Build the code via the (internal) ros build farm and create debian packages

jarkenau, 2023-01-12:
First steps:

- [ ] Research the required versions for gRPC, flatbuffers, protocol buffers, ...
- [ ] Check if the required versions are available for Ubuntu 20.04 LTS with `apt`, if not we need to use `git submodules`
- [ ] Check if the versions are in the [ROS Package index](https://github.com/ros/rosdistro/blob/97944875c68c878f3571abbeb5b8dac50d1a4e2a/rosdep/base.yaml#L7327) -> can be used via. `package.xml`

I will probably start by trying this for the ros-dumper.

jarkenau, 2023-01-12:
For integration in the CI we can use https://github.com/jspricke/ros-deb-builder-action, to build the debian packages.