DFKI-NI/seerep #183: Image and point cloud attributes defined in two places

jarkenau, 2022-09-14:
One in the core:

https://github.com/agri-gaia/seerep/blob/8d9cce7abd2a58a6352bd50ba7fde8191c8a0e8b/seerep-hdf5/seerep-hdf5-core/include/seerep-hdf5-core/hdf5-core-image.h#L38-L53

and once in the hdf5 access: 

https://github.com/agri-gaia/seerep/blob/8d9cce7abd2a58a6352bd50ba7fde8191c8a0e8b/seerep-hdf5/seerep-hdf5-fb/include/seerep-hdf5-fb/hdf5-fb-image.h#L43-L55

To-Do: unify into one place.

jarkenau, 2022-10-06:
@mhoellmann to which place should the definitions be moved? In the Core we have the benefit of using the definitions between fb and pb. But if we have fields which are only supported in one format, it could get confusing.

mhoellmann, 2022-10-07:
The definitions in the core should be used. The [hdf5-pb-image](https://github.com/agri-gaia/seerep/blob/8d9cce7abd2a58a6352bd50ba7fde8191c8a0e8b/seerep-hdf5/seerep-hdf5-pb/include/seerep-hdf5-pb/hdf5-pb-image.h) already uses the core definitions. It should be done similarly.

jarkenau, 2022-10-07:
It's now changed for point clouds, should also be done for images and points in fb

PR #190 (2022-10-10): update fb to use hdf5 definitions from seerep-hdf5-core
Since I just changed the locations of the hdf5 definitions, everything should work as bevor. 

Closes #183
closed 2022-10-10 by jarkenau