DFKI-NI/seerep #271: Unify import scheme in C++ for SEEREP components

jarkenau, 2023-04-03:
Some SEEREP components get imported with `#include "..."` while others use `#include <...>`. We should unify this into using the latter one everywhere.

Mark-Niemeyer, 2023-04-03:
Could you provide some examples?

jarkenau, 2023-04-03:
https://github.com/agri-gaia/seerep/blob/bb81519a0044140c340d8bb0be96e148d64234a9/seerep-hdf5/seerep-hdf5-py/include/seerep-hdf5-py/hdf5-py-pointcloud.h#L8
https://github.com/agri-gaia/seerep/blob/bb81519a0044140c340d8bb0be96e148d64234a9/seerep-hdf5/seerep-hdf5-py/include/seerep-hdf5-py/hdf5-py-tf.h#L8
https://github.com/agri-gaia/seerep/blob/bb81519a0044140c340d8bb0be96e148d64234a9/seerep-hdf5/seerep-hdf5-ros/include/seerep-hdf5-ros/hdf5-ros.h#L22