DFKI-NI/seerep #295: Bad lexical cast when checking for the camera intrinsic

jarkenau, 2023-06-02:
To my understanding, since PR #280, we require a camera intrinsic to be present in the server to be able to save an image. 
For that, the `uuid_camerainstrinsic` attribute was added in the `ìmage` message:
https://github.com/agri-gaia/seerep/blob/b3cd30a3e8866bf2ee87feddd0ee8e46d0b8a0c8/seerep_msgs/fbs/image.fbs#L18
But if the UUID is not set by a client sending data, the following cast will fail, since invalid memory will be accessed. Interestingly, no segmentation fault occurs. 
https://github.com/agri-gaia/seerep/blob/b3cd30a3e8866bf2ee87feddd0ee8e46d0b8a0c8/seerep_srv/seerep_core_fb/src/core_fb_image.cpp#L47
Therefore, please check if the attribute is set and then access it.

104H, 2023-06-07:
We disallowed the addition of new images without the presence of a valid Cam Intrinsics ID present. This field is mandatory and therefore should exist. I think this issue needs to be dealt with on the service definition layer.

Mark-Niemeyer, 2023-06-14:
In the most places I know we use the string_generator. I guess that throws an error it the string is unset and we do not have to check explicitly

e.g.
https://github.com/agri-gaia/seerep/blob/b3cd30a3e8866bf2ee87feddd0ee8e46d0b8a0c8/seerep_srv/seerep_core_fb/src/core_fb_image.cpp#L67-L68

PR #300 (2023-06-16): string gen instead of lexical cast camintrin uuid
Closes #295
closed 2023-06-16 by 104H