DFKI-NI/seerep #318: Start to refactor CMake

jarkenau, 2023-07-27:
A suitable quote that I found while figuring out (yet) another CMake Error:

> CMake is anarchy 

I think that also holds up for our project :disappointed: and we should start to refactor. For example:

The `seerep_core_fb` package is a pure CMake package, no catkin involved. We therefore configured it to be available via the `find_package()` CMake directive. A [...Config.cmake.in](https://github.com/agri-gaia/seerep/blob/main/seerep_srv/seerep_core_fb/cmake/SeerepCoreFbConfig.cmake.in) file is used to define its dependencies. This file is passed to CMake to generate the required `Config.cmake`, which is needed by `find_package`

The problem is that the `cmake.in` file has not been updated and still has Protocol Buffer  dependencies?!
https://github.com/agri-gaia/seerep/blob/0e4e058554a60eb00171d5e6d5ec07a5eb55980a/seerep_srv/seerep_core_fb/cmake/SeerepCoreFbConfig.cmake.in#L11-L29

Now if another package want's to use `seerep_core_fb` and the PB packages have not been build before, an error like this will be thrown

```bash
CMake Error at /seerep/devel/lib/cmake/seerepcorefb/SeerepCoreFbConfig.cmake:51 (find_package):
  By not providing "FindSeerepHdf5Pb.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "SeerepHdf5Pb", but CMake did not find one.

  Could not find a package configuration file provided by "SeerepHdf5Pb" with
  any of the following names:

    SeerepHdf5PbConfig.cmake
    seerephdf5pb-config.cmake
```

Of course, this doesn't happen if you run `catkin build` to build all packages.

Mark-Niemeyer, 2023-07-27:
extension of #141

jarkenau, 2023-07-27:
This issue probably also applies to other packages