DFKI-NI/seerep #328: Getting error while querying data from server

Qalababbass, 2023-08-03:
Hi , I am not able to query data from server after updating to latest update at main branch. I used the same method to send images to server and query them but now its giving me an error. So , i tried to run tutorial on the new update but still its not working. I am using grpc_fb_queryImage.py to query images

`Traceback (most recent call last):
  File "/seerep/src/examples/python/gRPC/images/gRPC_fb_queryImage.py", line 80, in <module>
    for responseBuf in stub.GetImage(bytes(buf)):
  File "/usr/local/lib/python3.8/dist-packages/grpc/_channel.py", line 475, in __next__
    return self._next()
  File "/usr/local/lib/python3.8/dist-packages/grpc/_channel.py", line 881, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.INVALID_ARGUMENT
        details = "bad optional access"
        debug_error_string = "UNKNOWN:Error received from peer ipv6:%5B::1%5D:9090 {grpc_message:"bad optional access", grpc_status:3, created_time:"2023-08-03T20:01:41.56895478+02:00"}"`

This is the output at terminal where server is running

`[2023-08-03 20:01:41.568641]<info>: sending images with this query parameters:
 time interval (1610549273/1938549273)
 labels generalcategory: 0; 'testlabel0' 'testlabelgeneral0' 
[2023-08-03 20:01:41.568689]<info>: loading image from images/
[2023-08-03 20:01:41.568803]<error>: bad optional access`

Any idea how to fix it.

jarkenau, 2023-08-07:
Which data is currently is the server? It might already be fixed in https://github.com/agri-gaia/seerep/commit/f0dc05aeaa8d0c504425a14f56a5eafa492a8da5 (although not merged into main). I'm currently not able to reproduce your error to confirm the fix.

Qalababbass, 2023-08-07:
I tried with rocas rgb image and also dummy image as in tutorial. I am getting this error in latest main as of on Friday, didnt checked if there's a new commit or not.
I have an old version of main and it works on that as normal.

Qalababbass, 2023-08-07:
I just cloned the latest main and error in fb_queryImage.py is still there.

Mark-Niemeyer, 2023-08-07:
fixed in #332 and #333