DFKI-NI/seerep #342: No check in place if camera intrinsics matrices have the correct layout

mmeijerdfki, 2023-09-29:
Neither the schema files for protobuf/flatbuffers nor the server check if the following transferred matrices have the right layout:

- distortion matrix
- intrinsic matrix
- rectification matrix
- projection matrix

This for example causes the server to crash randomly on my machine, when running the script [gRPC_pb_sendLabeledImageGrid.py](https://github.com/agri-gaia/seerep/blob/bug/camintrins-loading-projects/examples/python/gRPC/images/gRPC_pb_sendLabeledImageGrid.py) beforehand, then moving the created project out of the folder, restarting the server, moving the project file back and finally executing the script [gRPC_fb_loadProjects](https://github.com/agri-gaia/seerep/blob/bug/camintrins-loading-projects/examples/python/gRPC/meta/gRPC_fb_loadProjects.py). 
Note: these steps were done on the **bug/camintrins-loading-projects** branch.
The intrics matrix in the example only has three values.

The problem lies in here where the camera intrics matrix is used to calculate the frustum:
https://github.com/agri-gaia/seerep/blob/ea2f60522b305c271ba6056de0d790c1c9d035de/seerep_hdf5/seerep_hdf5_core/src/hdf5_core_image.cpp#L138-L139

Here the 5th value of the intrisics matrix is accessed, but not set by the example script when transferring the images. This causes undefined behaviour and the crash of the server after loading some images. The debugger shows that in this case eventually one image has it's `field_of_view_y` set to `inf`.

Possible solutions which came to my mind researching this problem could be:
1. set fixed sizes for the matrices in the schema files using flatbuffers structs and protobufs nested messages.
2. checking for the correct layout on the server side.

jarkenau, 2023-10-23:
I think checking for the correct size on the server size, makes the most sense in this case, as this would result in the same approach for FlatBuffers and Protocol Buffers. Flatbuffers has support fixed size arrays, when they are nested inside a `struct` so something like:

```
struct Vec9 {
    v:[float:9];
}

struct Vec12 {
    v:[float:12];
}

table CameraIntrinsics
{
    header:Header;
    height:uint;
    width:uint;
    distortion_model:string;
    distortion:[double]; // size depends on distortion_model
    intrinsic_matrix:Vec9; // 3x3 major row matrix
    rectification_matrix:Vec9; // 3x3 major row matrix
    projection_matrix:Vec12; // 3x4 major row matrix
    binning_x:uint;
    binning_y:uint;
    region_of_interest:RegionOfInterest;
    maximum_viewing_distance:double;
}
``` 
would be possible (based on the layout of the [ROS message](https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/CameraInfo.html)). However, as  ProtoBuf does not support fixed length arrays, we would have to check the size of the matrices ourselves anyway. Thus, we should not rely on any format specific feature.  

Defining a matrix via a nested message, using a repeated field, does not provide any benefit IMO, instead of using a related field of primitive types.

mmeijerdfki, 2023-10-23:
In protobuf fixed size arrays could be mimicked like this:

`syntax = "proto3";

package seerep.pb;

import "header.proto";
import "region_of_interest.proto";

message CameraIntrinsics
{
  message Vec9
  {
    required double v0 = 1;
    required double v1 = 2;
    required double v2 = 3;
    required double v3 = 4;
    required double v4 = 5;
    required double v5 = 6;
    required double v6 = 7;
    required double v7 = 8;
    required double v8 = 9;
  }
  message Vec12
  {
    required double v0 = 1;
    required double v1 = 2;
    required double v2 = 3;
    required double v3 = 4;
    required double v4 = 5;
    required double v5 = 6;
    required double v6 = 7;
    required double v7 = 8;
    required double v8 = 9;
    required double v9 = 10;
    required double v10 = 11;
    required double v11 = 12;
  }
  Header header = 1;
  uint32 height = 2;
  uint32 width = 3;
  string distortion_model = 4;
  repeated double distortion = 5; // size depends on distortion_model
  Vec9 intrinsic_matrix = 6; // 3x3 major row matrix
  Vec9 rectification_matrix = 7; // 3x3 major row matrix
  Vec12 projection_matrix = 8; // 3x4 major row matrix
  uint32 binning_x = 9;
  uint32 binning_y = 10;
  RegionOfInterest region_of_interest = 11;
  double maximum_viewing_distance = 12;
}
`
The nested messages are not a requirement, though I think it is a cleaner solution when the messages (Vec9, Vec12) don't have to be reused somewhere else. The required keywords could also be added to the matrices, so that the user has to provide them, otherwise the server has to make sure to set default values.

jarkenau, 2023-10-24:
Ok got it, two comments on this:
1. The `required` keyword has been removed in the `proto3` syntax, thus we can't rely on it.
2. Mimicking the fixed length arrays is possible, but it doesn't scale well with larger arrays or matrices. Further, the fields have to be set by name e.g `v[x]` which is quite unappealing to me.

mmeijerdfki, 2023-10-24:
Alright, I've mistakenly took the `required` keyword for being in the newer version as well and there are good reasons not to use it anyway.
Therefore, server-side checking has to be done in any case and I agree that redefining the `proto3` message is not the most elegant solution. I thought that the interface would be more restrictive and it is clearer which input has to be given, but there is still the problem of values being left out. This is at least the case with `proto3`.

On the other hand FlatBuffers fields offer the use of a `required` attribute. 
Though the most coherent solution would probably the one you already mentioned: to handle the checks only on the server-side and not modifying any of the message definitions.

Mark-Niemeyer, 2023-10-30:
Additionally, if the distortion model isn't set and the distortion matrix is unset, an exception is thrown:

https://github.com/agri-gaia/seerep/blob/1a83ae725bebf35ff3d14200fad2ea274e05732e/seerep_hdf5/seerep_hdf5_core/src/hdf5_core_cameraintrinsics.cpp#L26-L27

PR #360 (2024-01-26): Add check for the correct layout of camera intrinsics matrices.
Closes #342. This also corrects a typo in Frustum.

Edit:
Empty values are now allowed, except for the intrinsics matrix.
closed 2024-01-26 by Mark-Niemeyer